leonardt/fault #4: Add more backends

rsetaluri, 2018-07-19:
Just to note:
-- python simulator
-- system verilog

phanrahan, 2018-11-16:
When I try to run the python simulator I get an error.
```
Python simulator is not actively supported
```

rsetaluri, 2018-11-20:
There was an issue with passing in clock types to the python simulator. I will try to reproduce. But in general from discussions with @leonardt it seems magma is well upstream of the simulator, so there maybe several unsupported features.

leonardt, 2018-11-21:
Yes I think there were some non-trivial issues with getting some of the fault features supported in the Python simulator. We can remove this error if we decide we're going to reactive support for the python simulator, but then we'll need to revisit some of those problems. I don't remember exactly what the issues were off the top of my head, but I'm sure if we start using it we'll discover them.

One immediate one is a lack of support for tuples, which I think we are starting to use now in Garnet (and therefore fault). @shacklettbp was working on tuple support for a different flow through the simulator, so that support may be coming soon.

leonardt, 2018-12-14:
System verilog backend is now operational, python simulator warning has been removed. I'll add it to the magma simulator target test fixture. I'll open a separate branch/issue to track this.

closed 2018-12-14 by leonardt