leonardt/fault #8: Define the semantics of tester.poke,eval,expect,step

rsetaluri, 2018-07-20:
There seems to be some ambiguity in these commands. These should be well defined. Of course, their associated code is well defined but we should consider their semantics.

leonardt, 2018-07-20:
What we decided today: `expect` semantics should be that the expectation of the result after `eval` is called. Currently it is the expectation of the result *when* the next `eval` is called.

leonardt, 2018-07-20:
Unresolved issue is whether we need to call `eval` at all. One option is to have an `eval` implicit on every `poke`, and then do analysis to reason about when `eval` actually needs to be called (for optimization purposes).

We could provide two modes. One simple mode where `eval` is called on every poke, and an optimized mode where the user can control when `eval` occurs.

leonardt, 2018-08-27:
I think this is cleared up with the new Action pattern.

Before we close this issue, I think it would be good to write up some documentation on this.

rsetaluri, 2018-08-28:
It would be good to do so. However we define it should be agnostic of the target simulator, and our targets should always properly map our semantics to the target's semantics.

phanrahan, 2018-09-14:
Are these functions documented? I couldn't find them.

rsetaluri, 2018-09-14:
They are not yet documented. I can work on this. I think the most useful documentation is how they map to the backend simulators.

As a rough guideline, for verilator:
-- `poke(x, value)` --> `set_x(value)`
-- `expect(x, value)` --> `assert(get_x() == value)`
-- `eval()` --> `eval()`
-- `step(n)` -->
```
for i = 1 : n
    set_clk(~get_clk())
    eval()
```

leonardt, 2018-09-17:
Actions are now documented at https://github.com/leonardt/fault/blob/master/doc/actions.md, will open a separate issue for documenting `Tester`.

closed 2018-09-17 by leonardt