leonardt/fault #20: Type mismatch in test vector generation

rsetaluri, 2018-08-14:
On the following line it looks like we expect a bit vector type:

https://github.com/leonardt/fault/blob/154427413be890a4b60a8f94aeaaae37eb6eac95/fault/verilator_target.py#L82

But if you look at the following function, you see that test vectors are generated as integers. Is there some translation missing here?

https://github.com/leonardt/fault/blob/154427413be890a4b60a8f94aeaaae37eb6eac95/fault/test_vectors.py#L25

leonardt, 2018-08-14:
This is a migration artifact. 

Our verilator target has been updated to use the bitvector interface, while the old code used `int`s and lists of bools. We should update the test_vector generation logic to use BitVectors, but we'll also need to make sure the tests in mantle that use this logic still work. There are some tests that use the test vector generator for circuits and functions and just compare the results (so this currently works since they don't use the verilator target)

rsetaluri, 2018-08-15:
Solved by #21

closed 2018-08-15 by rsetaluri