leonardt/fault #23: Semantics mismatch between generate_function_test_vectors and backends

leonardt, 2018-08-17:
Those test vectors assume that the output is a function of the inputs (since this was used for testing combinational circuits). I'm not sure what the best resolution here. One is to have a version of the backends for testing combinational circuits, where `eval` occurs before the outputs are checked. I think this will be cleaned up with the new Action interface (peek, poke, expect), so perhaps we can just add a temporary workaround until that's ready.

rsetaluri, 2018-08-17:
I'd advocate for a temp fix. I think the refactor should clean up a lot of that stuff. My only worry is the way users of fault are using that stuff may not adhere to the new abstractions but I think we can make everything in sync after refactoring.

leonardt, 2018-08-17:
Temporary workaround with c67f7cfe1c4f423c07971db873842be209cb1eb1

Basically, it constructs the test vectors as before, then reorders them so that the result of the previous vector is expected (basically shifts the expectations up by one, and inserts an "Any" in the first vector).

closed 2018-08-17 by leonardt