leonardt/fault #55: Allow trace in verilator

Kuree, 2019-01-21:
in `verilator_utils.py`, function `verilator_cmd()` provides a way to construct verilator commands. I think it would be nice to have an option to enable trace, which will making debugging much easier.

leonardt, 2019-01-21:
There's support for using the `flags` parameter, here's an example

* flags declaration (list of strings): https://github.com/leonardt/fault/blob/6c1822c31044c9ddb1976411dccfaf93c374e581/tests/test_verilator_target.py#L42
* usage: https://github.com/leonardt/fault/blob/6c1822c31044c9ddb1976411dccfaf93c374e581/tests/test_verilator_target.py#L49-L51

I think this is useful enough that we can add a dedicated parameter (and documentation), so I'll leave this issue open until that's resolved

leonardt, 2019-01-21:
Also, if `--trace` is used, it will dump the output to a `.vcd` with the following ofrmat `<directory>/logs/{circuit_name}.vcd`, see https://github.com/leonardt/fault/blob/master/fault/verilator_target.py#L56-L57 for more info.

For more info on how trace is used, you can inspect the generated harness by fault. Generically, fault leverages the `VM_TRACE` variable and uses `#ifdef` statements in the generated code to insert tracing logic. Let me know how that works out for you and if you want any added functionality (e.g. some ability to extend the tracing behavior, particularly related to recording of time steps). We could look into adding an action explicitly for recording trace values (rather than dumping all values every eval/step).

Kuree, 2019-01-21:
Thanks!

I'd like to have a way to correlate action step `i` with the timestamp (`main_time`) in the VCD files. The generated driver files only shows the `i` with assert. It would nice to have a signal in VCD that indicates the steps.