leonardt/fault #61: Issue with setting nested arrays

rsetaluri, 2019-02-11:
```RecursionError: maximum recursion depth exceeded while calling a Python object
Exception ignored in: '_pydevd_frame_eval.pydevd_frame_evaluator_darwin_36_64.get_bytecode_while_frame_eval'
RecursionError: maximum recursion depth exceeded while calling a Python object
RecursionError: maximum recursion depth exceeded while calling a Python object
Exception ignored in: '_pydevd_frame_eval.pydevd_frame_evaluator_darwin_36_64.get_bytecode_while_frame_eval'
RecursionError: maximum recursion depth exceeded while calling a Python object
F
tests/helper_test_readyvalid.py:211 (test_clock_adjusted_2dlb_flicker_ce_with_2x2_stride)
def test_clock_adjusted_2dlb_flicker_ce_with_2x2_stride():
        c = coreir.Context()
        cirb = CoreIRBackend(c)
    
        testcircuit = DefineTwoDimensionalLineBuffer(cirb, Array(8, In(Bit)), 1, 1, 2, 2, 8, 8, 2, 2, 0, 0, True)
    
        tester = fault.Tester(testcircuit, testcircuit.CLK)
    
        for i in range(30):
            if i % 2 == 0:
>               tester.circuit.I[0][0] = int2seq(1, 8)
E               TypeError: 'PortWrapper' object does not support indexing

helper_test_readyvalid.py:222: TypeError```

leonardt, 2019-02-12:
#62 has a preliminary support for setting arrays (see this test https://github.com/leonardt/fault/pull/62/files#diff-be25ab5c75d3019d597f051e308a886cR84), let me know if this fixes the issue.

Can I check out the above test somewhere to try it out? (I'm guessing it's in aetherling, CCing @david-durst)

David-Durst, 2019-02-12:
https://github.com/David-Durst/aetherling/blob/2043af4641f5bf46f5dd111d0812acf98f4eb932/tests/helper_test_readyvalid.py#L19-L72

David-Durst, 2019-02-12:
Note, in order to run this, you'll need the david-ram-tests magma, mantle, aetherling and fault branches (at least as of 6 pm this evening)

leonardt, 2019-02-12:
I had to make the following changes (basically replace `int2seq` with `BitVector`).
```diff
diff --git a/tests/helper_test_readyvalid.py b/tests/helper_test_readyvalid.py
index 723cb17..cf5efe0 100644
--- a/tests/helper_test_readyvalid.py
+++ b/tests/helper_test_readyvalid.py
@@ -26,8 +26,8 @@ def test_2dlb_flicker_ce_with_2x2_stride():
 
     passes = ["rungenerators", "wireclocks-coreir", "verifyconnectivity --noclkrst", "flattentypes", "flatten", "verifyconnectivity --noclkrst", "deletedeadinstances"]
     namespaces = ["aetherlinglib", "commonlib", "mantle", "coreir", "global"]
-    #magma.compile(f"build/{testcircuit.name}", testcircuit, output="coreir-verilog",
-    #              passes=passes, namespaces=namespaces, context=c)
+    magma.compile(f"build/{testcircuit.name}", testcircuit, output="coreir-verilog",
+                  passes=passes, namespaces=namespaces, context=c)
 
     import fault
     tester = fault.Tester(testcircuit, testcircuit.CLK)
@@ -191,21 +191,21 @@ def test_any_type_ram_flicker_we():
 
     sim.set_value(testcircuit.WADDR[0], 0, scope)
     sim.set_value(testcircuit.RADDR[0], 0, scope)
-    sim.set_value(testcircuit.WDATA[0], int2seq(0, 8), scope)
-    sim.set_value(testcircuit.WDATA[1], int2seq(0, 8), scope)
-    sim.set_value(testcircuit.WDATA[2], int2seq(0, 8), scope)
-    sim.set_value(testcircuit.WDATA[3], int2seq(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[0], BitVector(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[1], BitVector(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[2], BitVector(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[3], BitVector(0, 8), scope)
     sim.set_value(testcircuit.WE, 0, scope)
     for i in range(20):
         sim.evaluate()
         if i == 0:
-            sim.set_value(testcircuit.WDATA[0], int2seq(1, 8), scope)
+            sim.set_value(testcircuit.WDATA[0], BitVector(1, 8), scope)
         if i == 2:
-            sim.set_value(testcircuit.WDATA[1], int2seq(1, 8), scope)
+            sim.set_value(testcircuit.WDATA[1], BitVector(1, 8), scope)
         if i == 16:
-            sim.set_value(testcircuit.WDATA[2], int2seq(1, 8), scope)
+            sim.set_value(testcircuit.WDATA[2], BitVector(1, 8), scope)
         if i == 18:
-            sim.set_value(testcircuit.WDATA[3], int2seq(1, 8), scope)
+            sim.set_value(testcircuit.WDATA[3], BitVector(1, 8), scope)
         sim.evaluate()
         sim.advance_cycle()
         sim.evaluate()
@@ -215,10 +215,10 @@ def test_any_type_ram_flicker_we():
     sim.evaluate()
     sim.advance_cycle()
     sim.evaluate()
-    sim.set_value(testcircuit.WDATA[0], int2seq(0, 8), scope)
-    sim.set_value(testcircuit.WDATA[1], int2seq(0, 8), scope)
-    sim.set_value(testcircuit.WDATA[2], int2seq(0, 8), scope)
-    sim.set_value(testcircuit.WDATA[3], int2seq(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[0], BitVector(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[1], BitVector(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[2], BitVector(0, 8), scope)
+    sim.set_value(testcircuit.WDATA[3], BitVector(0, 8), scope)
     print("RDATA: " + str([seq2int(sim.get_value(testcircuit.RDATA, scope)[r]) for r in range(4)]))
     sim.set_value(testcircuit.WE, 0, scope)
     sim.evaluate()
@@ -233,14 +233,14 @@ def test_clock_adjusted_2dlb_flicker_ce_with_2x2_stride():
 
     testcircuit = DefineTwoDimensionalLineBuffer(cirb, Array(8, In(Bit)), 1, 1, 2, 2, 8, 8, 2, 2, 0, 0, True)
 
-    tester = fault.tester(testcircuit, testcircuit.CLK)
+    tester = fault.Tester(testcircuit, testcircuit.CLK)
 
     for i in range(30):
[... 235 lines cut ...]
dict_keys(['AnyDimensionalLineBuffer_Array_8_In_Bit__type__1_1_pxPerClock__2_2_window__8_8_img__2_2_stride__0_0_origin_inst0', 'DelayedBuffer_Array_2_Array_2_Array_8_In_Bit____t_4n_1k_16emittingPeriod_0initialDelay_inst0', 'Counter1_Mod2CE_inst0', 'coreir_const10_inst0', 'coreir_eq_1_inst0', 'and_inst0', 'or_inst0', 'and_inst1', 'or_inst1', 'and_inst2']) O False
dict_keys(['AnyDimensionalLineBuffer_Array_8_In_Bit__type__1_1_pxPerClock__2_2_window__8_8_img__2_2_stride__0_0_origin_inst0', 'DelayedBuffer_Array_2_Array_2_Array_8_In_Bit____t_4n_1k_16emittingPeriod_0initialDelay_inst0', 'Counter1_Mod2CE_inst0', 'coreir_const10_inst0', 'coreir_eq_1_inst0', 'and_inst0', 'or_inst0', 'and_inst1', 'or_inst1', 'and_inst2']) O False
dict_keys(['AnyDimensionalLineBuffer_Array_8_In_Bit__type__1_1_pxPerClock__2_2_window__8_8_img__2_2_stride__0_0_origin_inst0', 'DelayedBuffer_Array_2_Array_2_Array_8_In_Bit____t_4n_1k_16emittingPeriod_0initialDelay_inst0', 'Counter1_Mod2CE_inst0', 'coreir_const10_inst0', 'coreir_eq_1_inst0', 'and_inst0', 'or_inst0', 'and_inst1', 'or_inst1', 'and_inst2']) O False
dict_keys(['AnyDimensionalLineBuffer_Array_8_In_Bit__type__1_1_pxPerClock__2_2_window__8_8_img__2_2_stride__0_0_origin_inst0', 'DelayedBuffer_Array_2_Array_2_Array_8_In_Bit____t_4n_1k_16emittingPeriod_0initialDelay_inst0', 'Counter1_Mod2CE_inst0', 'coreir_const10_inst0', 'coreir_eq_1_inst0', 'and_inst0', 'or_inst0', 'and_inst1', 'or_inst1', 'and_inst2']) I False
dict_keys(['AnyDimensionalLineBuffer_Array_8_In_Bit__type__1_1_pxPerClock__2_2_window__8_8_img__2_2_stride__0_0_origin_inst0', 'DelayedBuffer_Array_2_Array_2_Array_8_In_Bit____t_4n_1k_16emittingPeriod_0initialDelay_inst0', 'Counter1_Mod2CE_inst0', 'coreir_const10_inst0', 'coreir_eq_1_inst0', 'and_inst0', 'or_inst0', 'and_inst1', 'or_inst1', 'and_inst2']) O False
[... cut ...] nd_inst2']) O False
dict_keys(['AnyDimensionalLineBuffer_Array_8_In_Bit__type__1_1_pxPerClock__2_2_window__8_8_img__2_2_stride__0_0_origin_inst0', 'DelayedBuffer_Array_2_Array_2_Array_8_In_Bit____t_4n_1k_16emittingPeriod_0initialDelay_inst0', 'Counter1_Mod2CE_inst0', 'coreir_const10_inst0', 'coreir_eq_1_inst0', 'and_inst0', 'or_inst0', 'and_inst1', 'or_inst1', 'and_inst2']) O False
=================================== 1 failed, 6 deselected in 4.91 seconds ====================================
```

leonardt, 2019-02-12:
(Edited the above diff to simplify it, my search and replace touched other parts of the file that weren't of interest)