leonardt/fault #63: Ability To Load Results Back To Python

David-Durst, 2019-02-17:
I've modified fault (https://github.com/David-Durst/fault/tree/aetherling_debugging) to work with Aetherling (https://github.com/David-Durst/aetherling/blob/master/aetherling/helpers/fault_helpers.py) so that I can write port values out to disk and then load them back after running the tests. I am now able to write all ports values out to disk, load them after the test runs, and assert based on the Python data structure.

This is necessay for Aetherling to use fault. The current expects interface does not allow for conditional expects, ie only expect a port to have a value if another port has a value. Conditional expects are necessary for me to test the ready-valid interfaces in Aetherling. Having the entire logs in a Python data structure also makes debugging easier as I can more easily write results interpreters.

Can the interface to fault be modified to support this functionality in a less hacky way?

leonardt, 2019-02-19:
This is an important feature request, will think about the design and come up with some proposals.

I think having the ability to perform more complex operations with `peek` + `expect` could achieve this? Basically, the expected value would be a symbolic expression based on a `peek` value (the current value of an output).

David-Durst, 2019-02-19:
I think that should work. That would allow me to express my conditionals.

One nice thing about my current approach is that I can use my debugging tools to filter the results. The staged meta-programming makes this harder, as you have to go through multiple layers. Is there any way to keep IDE integration?