leonardt/fault #95: Fault Doesn't Respect Magma's "compile_dir"

David-Durst, 2019-05-14:
The below test succeeds when I run it from my IDE. My IDE runs pytest from the `tests/haskell` directory. 

https://github.com/David-Durst/aetherling/blob/master/tests/haskell/test_downsampleStencil.py#L176-L212

My compile_dir is set at https://github.com/David-Durst/aetherling/blob/master/conftest.py#L12

However, when I run pytests -s from my project's root directory, I get the below error. I verified that the verilog file is created each time the test is run in `tests/haskell/vBuild`. I think that verilator is looking for the vBuild directory at the root of my project.

```
Error: Cannot find file containing module: downsampleStencilChain1Per32_Circuit.v
%Error: This may be because there's no search path specified with -I<dir>.
%Error: Looked in:
%Error:       downsampleStencilChain1Per32_Circuit.v
%Error:       downsampleStencilChain1Per32_Circuit.v.v
%Error:       downsampleStencilChain1Per32_Circuit.v.sv
%Error:       obj_dir/downsampleStencilChain1Per32_Circuit.v
%Error:       obj_dir/downsampleStencilChain1Per32_Circuit.v.v
%Error:       obj_dir/downsampleStencilChain1Per32_Circuit.v.sv
%Error: Exiting due to 9 error(s)
%Error: Command Failed /usr/local/Cellar/verilator/4.010/bin/verilator_bin -Wall -Wno-INCABSPATH -Wno-DECLFILENAME --cc downsampleStencilChain1Per32_Circuit.v --exe downsampleStencilChain1Per32_Circuit_driver.cpp --top-module downsampleStencilChain1Per32_Circuit
F
==================================================================================== FAILURES =====================================================================================
_____________________________________________________________________ test_downsample_stencil_1_per_32_fault ______________________________________________________________________

    def test_downsample_stencil_1_per_32_fault():
        from .downsampleStencilChain1Per32 import c, downsampleStencilChain1Per32

        magma.compile("vBuild/" + downsampleStencilChain1Per32.name, downsampleStencilChain1Per32, output="coreir-verilog",
                      passes=["rungenerators", "wireclocks-coreir", "verifyconnectivity --noclkrst", "flattentypes", "flatten", "verifyconnectivity --noclkrst", "deletedeadinstances"],
                      namespaces=["aetherlinglib", "commonlib", "mantle", "coreir", "global"], context = c)

        tester = fault.Tester(downsampleStencilChain1Per32, downsampleStencilChain1Per32.CLK)

        tester.poke(downsampleStencilChain1Per32.valid_data_in, 1)
        tester.poke(downsampleStencilChain1Per32.ready_data_out, 1)
        tester.poke(downsampleStencilChain1Per32.CE, 1)

        # these check the outputs, as there is a delay between feeding inputs in and getting the results back
        cur_row_to_check = 0
        cur_col_to_check = 0
        successfully_checked_all_valid_outputs = False
        for row in range(num_rows+6):
            for col in range(0,num_cols,2):
                # a necessary adjustment as running tests for multiple clocks after inputting full image
                # to get rest of the outputs
                if row < num_rows:
                    tester.poke(downsampleStencilChain1Per32.I0, image_matrix[row][col])
                    tester.poke(downsampleStencilChain1Per32.I1, image_matrix[row][col+1])
                tester.eval()
                #tester.expect(downsampleStencilChain1Per32.ready_data_in, 1)
                tester.step(2)
                tester.eval()
                print_start_clock(tester)
                print_nd_bit_array_port(tester, downsampleStencilChain1Per32.valid_data_out, "valid")
                print_nd_int_array_port(tester, downsampleStencilChain1Per32.O0, "O0")
                print_end_clock(tester)
>       tester.compile_and_run(target="verilator", skip_compile=True, directory="vBuild/")

tests/haskell/test_downsampleStencil.py:208:
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _
../fault/fault/tester.py:170: in compile_and_run
    self.compile(target, **kwargs)
../fault/fault/tester.py:150: in compile
    self.targets[target] = self.make_target(target, **kwargs)
../fault/fault/tester.py:70: in make_target
    return VerilatorTarget(self._circuit, **kwargs)
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _

self = <fault.verilator_target.VerilatorTarget object at 0x10d142080>
circuit = downsampleStencilChain1Per32_Circuit = DefineCircuit("downsampleStencilChain1Per32_Circuit", "I0", Array[8, In(Bit)], ...ock_2x2window_4x4img_2x2stride_0x0origin_inst0.valid, downsampleStencilChain1Per32_Circuit.valid_data_out)
EndCircuit()
directory = 'vBuild/', flags = [], skip_compile = True, include_verilog_libraries = [], include_directories = [], magma_output = 'coreir-verilog', circuit_name = None
magma_opts = {}

    def __init__(self, circuit, directory="build/",
                 flags=[], skip_compile=False, include_verilog_libraries=[],
                 include_directories=[], magma_output="coreir-verilog",
                 circuit_name=None, magma_opts={}):
        """
        Params:
            `include_verilog_libraries`: a list of verilog libraries to include
            with the -v flag.  From the verilator docs:
                -v <filename>              Verilog library

            `include_directories`: a list of directories to include using the
            -I flag. From the the verilator docs:
                -I<dir>                    Directory to search for includes
        """
        super().__init__(circuit, circuit_name, directory, skip_compile,
                         include_verilog_libraries, magma_output, magma_opts)
        self.flags = flags
        self.include_directories = include_directories

        # Compile the design using `verilator`
        driver_file = self.directory / Path(f"{self.circuit_name}_driver.cpp")
        verilator_cmd = verilator_utils.verilator_cmd(
            self.circuit_name, self.verilog_file.name,
            self.include_verilog_libraries, self.include_directories,
            driver_file.name, self.flags)
        if self.run_from_directory(verilator_cmd):
>           raise Exception(f"Running verilator cmd {verilator_cmd} failed")
E           Exception: Running verilator cmd verilator -Wall -Wno-INCABSPATH -Wno-DECLFILENAME --cc downsampleStencilChain1Per32_Circuit.v  --exe downsampleStencilChain1Per32_Circuit_driver.cpp --top-module downsampleStencilChain1Per32_Circuit failed

../fault/fault/verilator_target.py:105: Exception
```

leonardt, 2019-05-14:
@David-Durst can you tryout the branch https://github.com/leonardt/fault/pull/96

It adds a similar option to fault for configuring the compile behavior relative to the calling file, see https://github.com/leonardt/fault/pull/96/files#diff-423890b7e3c2aa4791f9e203b7f194cbR13 for an example on how to use it (basically, add that command to your conftest).

Let me know if you run into issues with it

David-Durst, 2019-05-14:
That fixed my issue. Thanks!

closed 2019-05-14 by David-Durst