leonardt/fault #112: Tests cannot be run in parallel

cdonovick, 2019-06-02:
Running tests with pytest-parallel often fails `test_setattr_nested_arrays_by_element[verilator-None]` with an assertion error at `fault/verilator_target.py:384`.  Have hit same assertion error less frequently in `test_tester_poke_internal_register[verilator-None]` and  `test_setattr_nested_arrays_bulk[verilator-None]`.

cdonovick, 2019-06-02:
```
(env-fault) donovick@telescape:~/work/fault% pytest --workers=auto                    
================================ test session starts =================================
platform linux -- Python 3.7.3, pytest-4.6.0, py-1.8.0, pluggy-0.12.0                 
rootdir: /home/donovick/work/fault, inifile: setup.cfg                                
plugins: cov-2.7.1, parallel-0.0.9                                                    
collected 70 items                                                                    
pytest-parallel: 12 workers (processes), 1 test per worker (thread)                   
s.......s...s.......F........x........................................                
====================================== FAILURES ======================================
__________________ test_setattr_nested_arrays_bulk[verilator-None] ___________________
                                                                                      
target = 'verilator', simulator = None                                                
                                                                                      
    def test_setattr_nested_arrays_bulk(target, simulator):                           
        circ = common.TestNestedArraysCircuit                                         
        tester = Tester(circ)                                                         
        expected = []                                                                 
        val = [random.randint(0, (1 << 4) - 1) for _ in range(3)]                     
        tester.circuit.I = val                                                        
        tester.eval()                                                                 
        tester.circuit.O.expect(val)                                                  
>       run_test(target, simulator, tester)                                           
                                                                                      
tests/test_setattr_interface.py:85:                                                   
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
tests/test_setattr_interface.py:33: in run_test                                       
    tester.compile_and_run(target, **kwargs)                                          
fault/tester.py:205: in compile_and_run                                               
    self.run(target)                                                                  
fault/tester.py:191: in run                                                           
    self.targets[target].run(self.actions, self.verilator_includes)                   
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _                                                                                                                              
                                                                                      
self = <fault.verilator_target.VerilatorTarget object at 0x7f95dd640128>              
actions = [Poke(NestedArraysCircuit.I, Array([BitVector[4](10), BitVector[4](11), Bit$
ector[4](15)], 3)), Eval(), Expect(O, Array([BitVector[4](10), BitVector[4](11), BitV$
ctor[4](15)], 3))]                                                                    
verilator_includes = [], num_tests = 0, _circuit = None                               
                                                                                      
    def run(self, actions, verilator_includes=[], num_tests=0,                        
            _circuit=None):                                                           
        driver_file = self.directory / Path(f"{self.circuit_name}_driver.cpp")        
        # Write the verilator driver to file.                                         
        src = self.generate_code(actions, verilator_includes, num_tests,              
                                 _circuit)                                            
        with open(driver_file, "w") as f:                                             
            f.write(src)                                                              
        # Run a series of commands: run the Makefile output by verilator, and         
        # finally run the executable created by verilator.                            
        verilator_make_cmd = verilator_utils.verilator_make_cmd(                      
            self.circuit_name)                                                        
        assert not self.run_from_directory(verilator_make_cmd)                        
        assert not self.run_from_directory(                                           
>           f"/bin/bash -c \"set -e -o pipefail; ./obj_dir/V{self.circuit_name}"      
            f" | tee ./obj_dir/{self.circuit_name}.log\"")                            
E       AssertionError                                                                
                                                                                      
fault/verilator_target.py:384: AssertionError                                         
-------------------------------- Captured stdout call -------------------------------$
make: Leaving directory '/home/donovick/work/fault/build/obj_dir'                     
make: Leaving directory '/home/donovick/work/fault/build/obj_dir'                     
[... cut ...]
bc.py:48: UserWarning: DEPRECATION WARNING: Use of BitVectorT(value, size) is deprecat
ed                                                                                    
    warnings.warn('DEPRECATION WARNING: Use of BitVectorT(value, size) is deprecated')
                                                                                      
-- Docs: https://docs.pytest.org/en/latest/warnings.html                              
====== 1 failed, 65 passed, 3 skipped, 1 xfailed, 11 warnings in 11.80 seconds =======
```

cdonovick, 2019-06-02:
The captured IO changes between runs

leonardt, 2019-06-03:
Issue should be fixed by https://github.com/leonardt/fault/commit/209e55fb27f0d9627b3c936eec9ab3144566310c

The tests are supposed to be using a tempdir, but it was overwritten as `build` probably for local debugging and never switched back. This results in the parallel workers sharing the `build` directory which is most likely the source of the error.

Running the tests in `examples/` fails for a similar reason (the tests share the same build dir), but I think that's okay for now. If you want to run the test suite in parallel, it's probably best to skip the examples tests by pointing to the testdir directly `pytest --works=auto tests/`.

Let me know if you run into other issues.

closed 2019-06-03 by leonardt