leonardt/fault #155: How To Print Internal Signals

David-Durst, 2019-08-29:
The main readme says that you can work with internal signals. It provides an example for calling .expect.

How can I print the internal signals. The line https://github.com/David-Durst/aetherling/blob/26ccb851cf179ed8e45a0e21fc76c3b95c2085d7/tests/test_bitonic_sort.py#L28 fails:
```
test_bitonic_sort.py:33: 
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 
../aetherling/helpers/fault_helpers.py:54: in compile_and_run
    }, directory="vBuild/")
../../fault/fault/tester.py:248: in compile_and_run
    self._compile_and_run(target=target, **kwargs)
../../fault/fault/tester.py:233: in _compile_and_run
    self.run(target)
../../fault/fault/tester.py:221: in run
    target_obj.run(self.actions, self.verilator_includes)
../../fault/fault/verilator_target.py:498: in run
    _circuit)
../../fault/fault/verilator_target.py:465: in generate_code
    code = self.generate_action_code(i, action)
../../fault/fault/verilog_target.py:90: in generate_action_code
    return self.make_print(i, action)
../../fault/fault/verilator_target.py:259: in make_print
    action.ports)
../../fault/fault/verilator_target.py:258: in <genexpr>
    ports = ", ".join(f"top->{verilog_name(port.name)}" for port in
_ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ _ 

name = coreir_ult11_inst0.in0

    def verilog_name(name):
        if isinstance(name, m.ref.DefnRef):
            return str(name)
        if isinstance(name, m.ref.ArrayRef):
            array_name = verilog_name(name.array.name)
            return f"{array_name}_{name.index}"
        if isinstance(name, m.ref.TupleRef):
            tuple_name = verilog_name(name.tuple.name)
            return f"{tuple_name}_{name.index}"
>       raise NotImplementedError(name, type(name))
E       NotImplementedError: (coreir_ult11_inst0.in0, <class 'magma.ref.InstRef'>)

../../fault/fault/verilog_utils.py:13: NotImplementedError
```

David-Durst, 2019-08-30:
This is addressed by #157.

David-Durst, 2019-09-07:
https://github.com/David-Durst/aetherling/blob/82970b97ed959fe62d42778088953e21cdd072e9/tests/test_space_time/test_nested_counters.py#L16 fails with the following message:

```
E               AssertionError: Got non-zero return code: 2. 
E                ../NestedCounters_TSeq_3_1_Int__hasCEFalse_hasResetFalse_driver.cpp:8:10: fatal error: 'VNestedCounters_TSeq_3_1_Int__hasCEFalse_hasResetFalse_NestedCounters_TSeq_3_1_Int__hasCEFalse_hasResetFalse.h' file not found
E               #include "VNestedCounters_TSeq_3_1_Int__hasCEFalse_hasResetFalse_NestedCounters_TSeq_3_1_Int__hasCEFalse_hasResetFalse.h"
E                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
E               1 error generated.
E               make: *** [NestedCounters_TSeq_3_1_Int__hasCEFalse_hasResetFalse_driver.o] Error 1
```

What am I doing wrong?

leonardt, 2019-09-11:
Can you try calling compile and run with 
```
magma_opts={"verilator_debug": True}
```

Sorry this isn't documented but I believe this is needed so verilator will preserve the internal signals. I will update the documentation if it works.

David-Durst, 2019-09-11:
(aetherling) durst@DN0a22a9e8:~/dev/W17-8/aetherling$ verilator --version
Verilator 4.014 2019-05-08 rev UNKNOWN_REV