leonardt/fault #160: Expect Array Of Tuples Fails

David-Durst, 2019-09-20:
https://github.com/David-Durst/aetherling/blob/14ef425256d53f05f4f2edc853cdcd2783ad1099/aetherling/HaskellPrelude.py#L68 fails with 
```
Traceback (most recent call last):
  File "/Users/durst/dev/W17-8/aetherling/aetherling/HaskellPrelude.py", line 68, in <module>
    tester.circuit.O.expect(fault_output[f_clk])
  File "/Users/durst/dev/W17-8/fault/fault/wrapper.py", line 60, in expect
    select_path.tester.expect(select_path, value)
  File "/Users/durst/dev/W17-8/fault/fault/tester.py", line 173, in expect
    value = make_value(port, value)
  File "/Users/durst/dev/W17-8/fault/fault/value_utils.py", line 22, in make_value
    return make_array(switch.T, switch.N, value)
  File "/Users/durst/dev/W17-8/fault/fault/value_utils.py", line 52, in make_array
    raise NotImplementedError(T, N, value)
NotImplementedError: (Tuple(0: Array[8, In(Bit)], 1: Array[8, In(Bit)]), 4, [(0, 0), (1, 1), (2, 2), (3, 3)])
```

The issue appears to be with https://github.com/leonardt/fault/blob/7aa6353cbbb42c9d5f26fa2f8f848714f6a5ce0b/fault/value_utils.py#L24.

When I try manually creating the expects for each element of the tuple, I get:
```
Running command: make -C obj_dir -j -f VModule_2.mk VModule_2
<STDOUT>
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.014/share/verilator/include -I/usr/local/Cellar/verilator/4.014/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o Module_2_driver.o ../Module_2_driver.cpp
/usr/bin/perl /usr/local/Cellar/verilator/4.014/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VModule_2.cpp VModule_2_Module_2.cpp VModule_2_coreir_ugt__W8.cpp VModule_2_coreir_neg__W8.cpp VModule_2_coreir_mux__W8.cpp > VModule_2__ALLcls.cpp
/usr/bin/perl /usr/local/Cellar/verilator/4.014/share/verilator/bin/verilator_includer -DVL_INCLUDE_OPT=include VModule_2__Dpi.cpp VModule_2__Syms.cpp > VModule_2__ALLsup.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.014/share/verilator/include -I/usr/local/Cellar/verilator/4.014/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o VModule_2__ALLsup.o VModule_2__ALLsup.cpp
clang++  -I.  -MMD -I/usr/local/Cellar/verilator/4.014/share/verilator/include -I/usr/local/Cellar/verilator/4.014/share/verilator/include/vltstd -DVL_PRINTF=printf -DVM_COVERAGE=0 -DVM_SC=0 -DVM_TRACE=0 -faligned-new -fbracket-depth=4096 -Qunused-arguments -Wno-parentheses-equality -Wno-sign-compare -Wno-uninitialized -Wno-unused-parameter -Wno-unused-variable -Wno-shadow       -c -o VModule_2__ALLcls.o VModule_2__ALLcls.cpp
      Archiving VModule_2__ALL.a ...
ar r VModule_2__ALL.a VModule_2__ALLcls.o VModule_2__ALLsup.o
ranlib VModule_2__ALL.a
</STDOUT>
<STDERR>
../Module_2_driver.cpp:62:18: error: no member named 'O_0_0' in 'VModule_2'
  my_assert(top->O_0_0, 0 & 255, 2, "Module_2.O[0].0");
            ~~~  ^
../Module_2_driver.cpp:63:18: error: no member named 'O_0_1' in 'VModule_2'
  my_assert(top->O_0_1, 0 & 255, 3, "Module_2.O[0].1");
            ~~~  ^
../Module_2_driver.cpp:64:18: error: no member named 'O_1_0' in 'VModule_2'
  my_assert(top->O_1_0, 1 & 255, 4, "Module_2.O[1].0");
            ~~~  ^
../Module_2_driver.cpp:65:18: error: no member named 'O_1_1' in 'VModule_2'
  my_assert(top->O_1_1, 1 & 255, 5, "Module_2.O[1].1");
            ~~~  ^
../Module_2_driver.cpp:66:18: error: no member named 'O_2_0' in 'VModule_2'
  my_assert(top->O_2_0, 2 & 255, 6, "Module_2.O[2].0");
            ~~~  ^
../Module_2_driver.cpp:67:18: error: no member named 'O_2_1' in 'VModule_2'
  my_assert(top->O_2_1, 2 & 255, 7, "Module_2.O[2].1");
            ~~~  ^
../Module_2_driver.cpp:68:18: error: no member named 'O_3_0' in 'VModule_2'
  my_assert(top->O_3_0, 3 & 255, 8, "Module_2.O[3].0");
            ~~~  ^
../Module_2_driver.cpp:69:18: error: no member named 'O_3_1' in 'VModule_2'
  my_assert(top->O_3_1, 3 & 255, 9, "Module_2.O[3].1");
            ~~~  ^
8 errors generated.
make: *** [Module_2_driver.o] Error 1
</STDERR>
```

It appears that this second issue is due to the naming of the ports. The ports in Module_2.v are 'O_0__1', 'O_0___2', 'O_1__1', ....

```
module Module_2 (input CLK/*verilator public*/, output [7:0] O_0__0/*verilator public*/, output [7:0] O_0__1/*verilator public*/, output [7:0] O_1__0/*verilator public*/, output [7:0] O_1__1/*verilator public*/, output [7:0] O_2__0/*verilator public*/, output [7:0] O_2__1/*verilator public*/, output [7:0] O_3__0/*verilator public*/, output [7:0] O_3__1/*verilator public*/, input [7:0] hi_0/*verilator public*/, input [7:0] hi_1/*verilator public*/, input [7:0] hi_2/*verilator public*/, input [7:0] hi_3/*verilator public*/);
```

David-Durst, 2019-09-30:
I'm trying to use this feature. However, I'm getting a bug that I thought you addressed.
I get the below error when running file: https://github.com/David-Durst/aetherling/blob/ec484351fba58006cf535d1dba7746e3e52722e0/aetherling/examples/ae_circuit13356-483.py#L88

```
../Module_3_driver.cpp:68:36: error: ‘class VModule_3’ has no member named ‘outoutout_0__0’; did you mean ‘outoutout_0___05F0’?
   printf("output 0.t0: %d\n", top->outoutout_0__0);
                                    ^~~~~~~~~~~~~~
                                    outoutout_0___05F0

```

How do I get the '05F' string inserted?

leonardt, 2019-09-30:
Looks like the processing of `__` (special symbol used by verilator which needs to be appended with `05F` to avoid symbol conflict) was missed in a code path, should be fixed by https://github.com/leonardt/fault/pull/165, let me know how it goes

David-Durst, 2019-09-30:
Fixed. Thanks!

closed 2019-09-30 by David-Durst