leonardt/fault #161: Print Doesn't Emit stdout When Running From Terminal

David-Durst, 2019-09-24:
[The file](https://github.com/David-Durst/aetherling/blob/master/aetherling/examples/ae_circuit12687-33.py) will emit the following output when run in my IDE.

```
Running command: ./obj_dir/VModule_1
<STDOUT>
clk: 0
clk: 1
</STDOUT>
<STDERR>

Got      : 0x0
Expected : 0x2
i        : 8
Port     : Module_1.O
</STDERR>
```

I believe this is written to stdout.

However, when I run the file from the command line with command `python ae_circuit12687-33.py`, I don't get this output. I only get the stderr. 

Am I running it wrong?

sgherbst, 2019-09-24:
Hi David, thanks for reporting this.  I just checked out the aetherling repo and tried running ``python ae_circuit12687-33.py`` in the directory **aetherling/aetherling/examples**.  Unfortunately I can't replicate the error -- here's what I see.
```text
Running command: ./obj_dir/VModule_1
<STDOUT>
clk: 0
clk: 1
</STDOUT>
<STDERR>

Got      : 0x0
Expected : 0x2
i        : 8
Port     : Module_1.O
</STDERR>
Found 1 error(s):
1) Got return code 1.
```
More information:
* Using python 3.7.4
* Latest version of fault from master branch (rev 9c5a70bf5d1526566ed84151f6c7a56949ab4172)
* magma, mantle, coreir installed from PyPI
    * magma-lang==1.0.18
    * mantle==1.0.8
    * coreir==2.0.29

David-Durst, 2019-09-25:
My apologies. The issues was due to an old fault version. I have multiple virtual environments on my laptop. My shell's environment still had fault 2.0.15 from PyPI while my IDE's environment has the latest fault from source.

closed 2019-09-25 by David-Durst