leonardt/fault #172: Fault Fails If Top Module Name Too Long

David-Durst, 2019-10-21:
https://github.com/David-Durst/aetherling/blob/185c5cf0ef85845b29ebf5f30912520c0c267a49/tests/test_space_time/test_reduce.py#L28-L46 passes.

However, it fails if you remove the call to `wrap_module_with_top`, which just wraps the module with one whose name is "top"

The failure message is that it can't find the top module. I believe this is because the name is too long. The error message is below

```
Running command: verilator -Wall -Wno-INCABSPATH -Wno-DECLFILENAME -Wno-fatal --cc Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit____.v --exe Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit_____driver.cpp --top-module Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit____
<STDERR>
%Error: Specified --top-module 'Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit____' was not found in design.
%Error: Exiting due to 1 error(s)
%Error: Command Failed /usr/local/Cellar/verilator/4.014/bin/verilator_bin -Wall -Wno-INCABSPATH -Wno-DECLFILENAME -Wno-fatal --cc Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit____.v --exe Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit_____driver.cpp --top-module Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit____
</STDERR>
```

leonardt, 2019-10-21:
This may be a verilator problem rather than a fault problem, I'm not sure if fault can do anything about this.

Can you check if the module `Reduce_S_n4_opNativeMapParallel_n1_opAdd_Atom_I_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit____O_Array_8_Out_Bit____I_Array_1_Tuple_0_Array_8_In_Bit___1_Array_8_In_Bit_____O_Array_1_Array_8_Out_Bit____` is indeed present in the output verilog files? If not, it may be an issue with the module naming rewriting logic

hofstee, 2019-10-21:
Doesn't verilator have issues with double underscores in names? Try removing all consecutive underscores and seeing if that helps.

leonardt, 2019-10-21:
Yea, we try to handle that with https://github.com/leonardt/fault/blob/master/fault/verilog_utils.py#L23-L28  where double underscores are replaced with ___05F, but there may be a place where we're missing this logic