leonardt/fault #173: Fault Failing To Find A Port

David-Durst, 2019-10-25:
The https://github.com/David-Durst/aetherling/blob/03141ab194035ed03c68f4823379c9b890f0c74f/aetherling/examples/ae_circuit5277-9.py fails with the following error. However, https://github.com/David-Durst/aetherling/commit/af0aea1b962669b4d7fa8bf0ed49f41e39d06d17 fixes the error.
```
<STDERR>
../top_driver.cpp:77:23: error: no member named 'O_0' in 'Vtop'; did you mean 'I_0'?
  printf("%d, ", top->O_0);
                      ^~~
                      I_0
./Vtop.h:31:12: note: 'I_0' declared here
    VL_IN8(I_0,7,0);
           ^
../top_driver.cpp:78:23: error: no member named 'O_1' in 'Vtop'; did you mean 'I_1'?
  printf("%d, ", top->O_1);
                      ^~~
                      I_1
./Vtop.h:32:12: note: 'I_1' declared here
    VL_IN8(I_1,7,0);
           ^
../top_driver.cpp:79:23: error: no member named 'O_2' in 'Vtop'; did you mean 'I_2'?
  printf("%d, ", top->O_2);
                      ^~~
                      I_2
```

leonardt, 2019-10-25:
The issue is that fault does not flatten `Array[N, Bit]` into an array of single bit values, instead it preserves the structure as a "bit vector".

To make this work, I looked at https://github.com/David-Durst/aetherling/blob/03141ab194035ed03c68f4823379c9b890f0c74f/aetherling/helpers/fault_helpers.py#L69 and saw that num_nested_layers was 1.  Changing it to 0 for this case caused the test bench to be generated properly (but the test itself failed).

I made the following change to the test code too: 
```python
❯ git diff
diff --git a/aetherling/examples/ae_circuit5277-9.py b/aetherling/examples/ae_circuit5277-9.py
index f22fc41..4229a2c 100644
--- a/aetherling/examples/ae_circuit5277-9.py
+++ b/aetherling/examples/ae_circuit5277-9.py
@@ -77,11 +77,11 @@ if __name__ == '__main__':
         if f_clk > 4:
             output_counter += 1
         tester.print("O: ")
-        fault_helpers.print_nested_port(tester, tester.circuit.O, num_nested_space_layers(ST_SSeq(4, ST_Bit())))
+        fault_helpers.print_nested_port(tester, tester.circuit.O, 0)
         tester.print("\n")
         if f_clk >= 4:
             tester.circuit.valid_down.expect(1)
         if f_clk >= 4 and fault_output_valid[output_counter]:
-            fault_helpers.expect_nested_port(tester, tester.circuit.O, fault_output[output_counter], num_nested_space_layers(ST_SSeq(4, ST_Bit())), 0)
+            fault_helpers.expect_nested_port(tester, tester.circuit.O, fault_output[output_counter], 0, 0)
         tester.step(2)
     fault_helpers.compile_and_run(tester)
```

Is it possible for your code to check if the leaf element is an Array of Bits, if so, set the nesting layer to 0? The other option is to have fault flatten this array of bits out, but this may be less efficient.

David-Durst, 2019-10-25:
David-Durst/aetherling@af0aea1 did essentially that. It wrapped Aetherling's bit in an Array[1,]. I'll stick with that fix.

closed 2019-10-25 by David-Durst