leonardt/fault #174: Error when trying to use fault installed from pip on kiwi

dillonhuff, 2019-10-28:
@leonardt I'm trying to create some tests of Verilog circuits created by Halide-to-Hardware for @David-Durst s upcoming PLDI submission.

When I try to run the following python script:
```python
import operator
import fault
```

I get the error:
```bash
(venv) dhuff@kiwi:~/embeddedHaskellAetherling/test/verilog_examples/halide_to_hardware/map$ python fault_tb.py 
Traceback (most recent call last):
  File "fault_tb.py", line 2, in <module>
    import fault
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/fault/__init__.py", line 1, in <module>
    from .real_type import RealIn, RealOut, RealKind, RealType
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/fault/real_type.py", line 1, in <module>
    from magma.port import Port, INPUT, OUTPUT, INOUT
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/magma/__init__.py", line 55, in <module>
    from .compile import *
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/magma/compile.py", line 10, in <module>
    from .uniquification import uniquification_pass, UniquificationMode
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/magma/uniquification.py", line 1, in <module>
    from dataclasses import dataclass
ModuleNotFoundError: No module named 'dataclasses'
```

I am in a virtual environment on kiwi, using python 3.6.8. I installed magma and fault using pip.

Have you seen this error before?

leonardt, 2019-10-28:
dataclasses is new in python3.6, you could try upgrading if that's easy or perhaps this backports pip module may work for you https://pypi.org/project/dataclasses/

leonardt, 2019-10-28:
(sorry Python 3.7 not 3.6)

dillonhuff, 2019-10-28:
Thanks lenny. That package fixed the dataclasses problem. Now it seems that I need to install coreir and make it visible to magma. I've tried to follow the install instructions for coreir linked to by magma, but I do not have sudo access on kiwi (which the install script requires).

Do I actually need to install coreir to ```/usr/bin/``` in order to fix this problem? The error I see now is below:

```python
(venv) dhuff@kiwi:~/embeddedHaskellAetherling/test/verilog_examples/halide_to_hardware/map$ python fault_tb.py 
Traceback (most recent call last):
  File "fault_tb.py", line 2, in <module>
    import fault
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/fault/__init__.py", line 3, in <module>
    from .tester import Tester
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/fault/tester.py", line 5, in <module>
    from fault.magma_simulator_target import MagmaSimulatorTarget
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/fault/magma_simulator_target.py", line 4, in <module>
    from magma.simulator.coreir_simulator import CoreIRSimulator
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/magma/simulator/coreir_simulator.py", line 5, in <module>
    from ..backend import coreir_
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/magma/backend/coreir_.py", line 11, in <module>
    import coreir
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/coreir/__init__.py", line 5, in <module>
    from coreir.global_value import COREGlobalValue_p
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/coreir/global_value.py", line 1, in <module>
    from .base import CoreIRType
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/coreir/base.py", line 2, in <module>
    from .lib import libcoreir_c
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/coreir/lib.py", line 27, in <module>
    libcoreir_c = load_coreir_lib("c")
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/coreir/lib.py", line 24, in load_coreir_lib
    return load_shared_lib('libcoreir-{}'.format(suffix))
  File "/home/dhuff/embeddedHaskellAetherling/venv/lib/python3.6/site-packages/coreir/lib.py", line 20, in load_shared_lib
    return cdll.LoadLibrary(libpath)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 426, in LoadLibrary
    return self._dlltype(name)
  File "/usr/lib/python3.6/ctypes/__init__.py", line 348, in __init__
    self._handle = _dlopen(self._name, mode)
OSError: libcoreir-c.so: cannot open shared object file: No such file or directory

```

leonardt, 2019-10-29:
You don't have to install coreir into /usr/bin, but you'll need the binary in your $PATH and the shared libraries in your $LD_LIBRARY_PATH, so
```
export PATH=/path/to/your/coreir/build/bin:$PATH
export LD_LIBRARY_PATH=/path/to/your/coreir/build/lib:$LD_LIBRARY_PATH
```

closed 2019-11-04 by dillonhuff