leonardt/fault #177: VCS Backend Fails

David-Durst, 2019-12-01:
The VCS backend fails with the tests in https://github.com/David-Durst/aetherling/blob/master/tests/test_space_time/test_downsample.py and https://github.com/David-Durst/aetherling/blob/master/tests/test_space_time/test_fifo.py.  [Here are the logs from test_fifo.py.](https://github.com/leonardt/fault/files/3907708/fifo.txt) Note: be sure to use the aetherling branch `vcs`.

I am running VCS on kiwi. I'm loading it with the command `module load vcs`.

I can't run verilator on kiwi because the version is too old (3.916), but the tests pass when run with verilator (4.020) on my local machine. I can't install verilator because Kiwi's ubuntu is missing required packages. 

I have asked for upgraded verilator to confirm

leonardt, 2019-12-04:
The compilation issue was a regression introduced in verilogAST introduced by https://github.com/leonardt/verilogAST-cpp/pull/20

This code path was disabled by https://github.com/leonardt/verilogAST-cpp/pull/25

can you try updating your coreir/verilogAST build?
```
cd /to/where/you/built/coreir
cd build/verilogAST-src/
git pull
cd ..
make
```

The tests now run for me without error from the verilog compiler, but the test benches themselves fail since they produce errors (is that expected?)

David-Durst, 2019-12-04:
I'm running on a separate branch of CoreIR (https://github.com/rdaly525/coreir/tree/update-json). Can I update verilogAST without updating CoreIR?

The test is expected to pass. It passes in verilator on my machine.

David-Durst, 2019-12-04:
I updated just verilogAST, now I get errors. It looks like no data is coming out. 
```
VCD+ Writer O-2018.09-SP1_Full64 Copyright (c) 1991-2018 by Synopsys Inc.
clk: 0
clk: 1
clk: 2
Error: "/home/durst/ae/aetherling/aetherling/helpers/vBuild/FIFO_tTSeq_4_2_Int_$
delay2_hasCETrue_hasResetFalse_hasValidTrue_tb.sv", 97: FIFO_tTSeq_4_2_Int__del$
y2_hasCETrue_hasResetFalse_hasValidTrue_tb: at time 120 ns
Failed on action=30 checking port valid_down. Expected 1, got 0
Error: "/home/durst/ae/aetherling/aetherling/helpers/vBuild/FIFO_tTSeq_4_2_Int__
delay2_hasCETrue_hasResetFalse_hasValidTrue_tb.sv", 101: FIFO_tTSeq_4_2_Int__del
ay2_hasCETrue_hasResetFalse_hasValidTrue_tb: at time 120 ns
Failed on action=31 checking port O. Expected 02, got 00
```

leonardt, 2019-12-05:
Upon inspecting the generated waveform, it looks like the clock is never ticking.

[image]

I suspect this is because the test bench assumes the lcock is initialized to 0 (as it is in verilator), but this is not the case in SV, sorry for this confusion, we are going to require that the clock is initialized in the future (generally in SV you can't assume anything about the intial values of signals).

I'll try updating the TB and see if that resolves the issue.

leonardt, 2019-12-05:
Just pushed a fix https://github.com/David-Durst/aetherling/commit/4c6f4b909ee2ebd37a82d48472d5782b6658502a the tests are now passing for me.

leonardt, 2019-12-05:
Woops, used inconsistent initial values, fixed in https://github.com/David-Durst/aetherling/commit/e16b8cc8cd2b9ad3840d1f2931bdafe00a45bd65

closed 2019-12-05 by leonardt