leonardt/fault #183: tests fail

phanrahan, 2019-12-10:
Lots of tests fail for me. Not sure what changed.
```
% pytest -s examples/test_simple_alu.py 
/Users/hanrahan/miniconda3/lib/python3.7/site-packages/pysmt/walkers/generic.py:43: DeprecationWarning: Using or importing the ABCs from 'collections' instead of from 'collections.abc' is deprecated since Python 3.3,and in 3.9 it will stop working
  if len(nodetypes) == 1 and isinstance(nodetypes[0], collections.Iterable):
/Users/hanrahan/miniconda3/lib/python3.7/site-packages/ast_tools/immutable_ast.py:10: UserWarning: /Users/hanrahan/miniconda3/lib/python3.7/site-packages/ast_tools/immutable_ast.py generated for (3, 6)does not match system version (3, 7)
  warnings.warn(f"{__file__} generated for (3, 6)"
============================= test session starts ==============================
platform darwin -- Python 3.7.5, pytest-5.3.1, py-1.8.0, pluggy-0.13.1
rootdir: /Users/hanrahan/git/fault, inifile: setup.cfg
plugins: astropy-header-0.1.1, arraydiff-0.3, codestyle-2.0.0, remotedata-0.3.2, openfiles-0.4.0, profiling-1.7.0, doctestplus-0.5.0
collected 7 items                                                              

examples/test_simple_alu.py /Users/hanrahan/git/coreir/src/binary/coreir.cpp:188 Running Runningvpasses
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:197 Running vpasses
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:238 Modified?: Yes
./Users/hanrahan/git/coreir/src/binary/coreir.cpp:188 Running Runningvpasses
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:197 Running vpasses
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:238 Modified?: Yes
./Users/hanrahan/git/coreir/src/binary/coreir.cpp:188 Running Runningvpasses
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:197 Running vpasses
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:238 Modified?: Yes
./Users/hanrahan/git/coreir/src/binary/coreir.cpp:188 Running Runningvpasses
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:197 Running vpasses
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:238 Modified?: Yes
./Users/hanrahan/git/coreir/src/binary/coreir.cpp:188 Running Runningvpasses
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:197 Running vpasses
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:238 Modified?: Yes
./Users/hanrahan/git/coreir/src/binary/coreir.cpp:188 Running Runningvpasses
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:10 In Run Generators
/Users/hanrahan/git/coreir/src/passes/transform/rungenerators.cpp:26 Done running generators
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:197 Running vpasses
/Users/hanrahan/git/coreir/src/binary/coreir.cpp:238 Modified?: Yes
Running command: verilator -Wall -Wno-INCABSPATH -Wno-DECLFILENAME -Wno-fatal --cc SimpleALU.v --exe SimpleALU_driver.cpp --top-module SimpleALU
<STDERR>
%Error: SimpleALU.v:39: syntax error, unexpected IDENTIFIER, expecting ')' or ','
module commonlib_muxn__N2__width2 (input [1:0] \in_data_0/*verilator public*/ , input [1:0] \in_data_1/*verilator public*/ , input [0:0] \in_sel/*verilator public*/ , output [1:0] \out/*verilator public*/ );
                                                                     ^~~~~~
%Error: SimpleALU.v:41: syntax error, unexpected IDENTIFIER
coreir_mux #(.width(2)) _join(.in0(in_data_0), .in1(in_data_1), .out(_join_out), .sel(in_sel[0]));
^~~~~~~~~~
%Error: SimpleALU.v:45: syntax error, unexpected IDENTIFIER, expecting ')' or ','
module commonlib_muxn__N2__width16 (input [15:0] \in_data_0/*verilator public*/ , input [15:0] \in_data_1/*verilator public*/ , input [0:0] \in_sel/*verilator public*/ , output [15:0] \out/*verilator public*/ );
                                                                       ^~~~~~
%Error: SimpleALU.v:47: syntax error, unexpected IDENTIFIER
coreir_mux #(.width(16)) _join(.in0(in_data_0), .in1(in_data_1), .out(_join_out), .sel(in_sel[0]));
^~~~~~~~~~
%Error: SimpleALU.v:51: syntax error, unexpected IDENTIFIER, expecting ')' or ','
module commonlib_muxn__N4__width16 (input [15:0] \in_data_0/*verilator public*/ , input [15:0] \in_data_1/*verilator public*/ , input [15:0] \in_data_2/*verilator public*/ , input [15:0] \in_data_3/*verilator public*/ , input [1:0] \in_sel/*verilator public*/ , output [15:0] \out/*verilator public*/ );
                                                                       ^~~~~~
%Error: SimpleALU.v:57: syntax error, unexpected IDENTIFIER
coreir_mux #(.width(16)) _join(.in0(muxN_0_out), .in1(muxN_1_out), .out(_join_out), .sel(in_sel[1]));
^~~~~~~~~~
%Error: SimpleALU.v:65: syntax error, unexpected IDENTIFIER, expecting ')' or ','
module Mux4xOutUInt16 (input [15:0] \I0/*verilator public*/ , input [15:0] \I1/*verilator public*/ , input [15:0] \I2/*verilator public*/ , input [15:0] \I3/*verilator public*/ , output [15:0] \O/*verilator public*/ , input [1:0] \S/*verilator public*/ );
                                                   ^~~~~~
%Error: SimpleALU.v:67: syntax error, unexpected IDENTIFIER
commonlib_muxn__N4__width16 coreir_commonlib_mux4x16_inst0(.in_data_0(I0), .in_data_1(I1), .in_data_2(I2), .in_data_3(I3), .in_sel(S), .out(coreir_commonlib_mux4x16_inst0_out));
^~~~~~~~~~~~~~~~~~~~~~~~~~~
%Error: SimpleALU.v:71: syntax error, unexpected IDENTIFIER, expecting ')' or ','
[... 182 lines cut ...]
            if chk_ret_code and returncode:
                err_msg += [f'Got return code {returncode}.']
    
            # look for errors in STDOUT or STDERR
            if err_str is not None:
                if err_str in stdout:
                    err_msg += [f'Found "{err_str}" in STDOUT.']
                if err_str in stderr:
                    err_msg += [f'Found "{err_str}" in STDERR.']
    
        # if any errors were found, print out STDOUT and STDERR if they haven't
        # already been printed, then print out what the error(s) were and
        # raise an exception
        if len(err_msg) != 0:
            display.error_printing()
            print(RED + BRIGHT + f'Found {len(err_msg)} error(s):' + RESET_ALL)
            for k, e in enumerate(err_msg):
                print(RED + BRIGHT + f'{k+1}) {e}' + RESET_ALL)
>           raise AssertionError
E           AssertionError

fault/subprocess_run.py:142: AssertionError
========================= 1 failed, 6 passed in 9.29s ==========================
% 
```

leonardt, 2019-12-10:
The error messages look related to a bug that was fixed by https://github.com/rdaly525/coreir/pull/820

Can you try updating your coreir/verilogAST installation?

```
# update verilogAST source
cd /path/to/coreir/build/verilogAST-src
git pull
cd ..
# update coreir
git pull
# rebuild/install
make && make install
``

leonardt, 2019-12-10:
Actually I was able to reproduce this, I think this was another regression introduced related to that original issue, looking into it

phanrahan, 2019-12-10:
I did update coreir and the problem persisted.

leonardt, 2019-12-10:
Should be fixed by https://github.com/rdaly525/coreir/pull/821

phanrahan, 2019-12-10:
I just pulled the latest release of coreir and installed. Tests still fail for the same reason. Is there something else I ned to do?

leonardt, 2019-12-10:
@phanrahan that branch hasn't been merged yet, waiting for a review, if you want to test the fix, you can check out the branch `hotfix-debug-port` in coreir and install it

leonardt, 2019-12-10:
Branch has been merged/released, @phanrahan can you test it again with the latest coreir?

phanrahan, 2019-12-10:
Works for me now.

closed 2019-12-10 by leonardt