leonardt/fault #186: Unable to parse DPI calls

Kuree, 2019-12-11:
Error:
```
    def _parse_error(self, msg, coord):
>       raise ParseError("%s: %s" % (coord, msg))
E       pyverilog.vparser.plyparser.ParseError: :1: before: import
```

SV:
```
import "DPI-C" function void breakpoint_trace(input int instance_id, input int stmt_id);
```

Kuree, 2019-12-11:
This is not a high priority since I've found ways to bypass the parsing verilog by directly exposing module interface to magma. 

Maybe it's a long-term goal to replace `pyverilog` with more advanced SV parser such as `slang`.

leonardt, 2019-12-11:
@Kuree do you have any examples using slang to parse verilog via a Python API? I don't think our use of pyverilog is very advanced (we just parse the module name/interface), so migrating to a new parser should be quick/easy

Kuree, 2019-12-11:
I only used `slang` in the context of C++ and it's fairly easy to use. I think it's straight-froward to create bindings in pybind, though requires some work. One major concern is that `slang` uses pretty recent C++ compiler and get it working on mac-os could be a pain.

Looking at other open source SV parser, I think we could also try out `verible` from Google. For completeness, here is the SV parser test suite:
https://symbiflow.github.io/sv-tests/

`slang` is in par with `verilator` but much faster. I haven't used `verible` though.