leonardt/fault #204: How To Call Expect For Long Runs

David-Durst, 2020-01-26:
I'm trying to simulate a 2D convolution for 1920x1080 and 640x480 images. This requires 2073600 example inputs and outputs for the 1920x1080 image and 307200 inputs and outputs for the 640x480 image. This many expect calls causes Verilator to crash because the c++ file is so large. My Chisel backend runs successfully because it can lazily load each input and output from a file on the necessary clock cycle. Fault cannot currently do this as the staged metaprogramming approach puts all inputs into a c++ file.

Can you add a call to load inputs and outputs from a file?

leonardt, 2020-01-26:
Yes, this is used for the garnet TBG.

Here's a simple test showing the feature: https://github.com/leonardt/fault/blob/master/tests/test_tester.py#L478-L515

Here's how it's used in that garnet TBG: https://github.com/StanfordAHA/garnet/blob/master/tbg.py#L174-L239

Let me know if you have specific questions

The main things to know are:
* `tester.file_open` opens a file and returns a handle
* `tester.file_read` reads a value from a file given a handle (usually done inside a loop).  You can use the chunk_size parameter to `file_open` to configure the size of each read
* `loop.file_write` writes data to a file (again usually done inside a loop)

leonardt, 2020-01-29:
@David-Durst have you had a chance to try this out? Let me know if you need help or if the features don't cover your use case

David-Durst, 2020-01-29:
@leonardt I haven't gotten to it yet. This looks like it could solve my use case, but I'll need to rewrite how I do my testing. I don't know how much of the file to read for each port for each time step, since I'm going to be assigning to nested ports each function call and the nested expect code is my own, not faults https://github.com/David-Durst/aetherling/blob/master/aetherling/helpers/fault_helpers.py#L90-L102