leonardt/fault #232: Add analog type to magma for fault mixed signal types

leonardt, 2020-04-04:
Related discussion https://github.com/leonardt/fault/pull/231

rsetaluri, 2020-04-04:
I think the type hierarchy should look like:

- Signal
  * Digital (for backward compat)
    - Bit
    - Clock
    - ...
  * Real
  * Elect
  * ...

I think we should move away from the pattern of `isisntance(x, Digital)` and have generic functions, like `is_digital()` etc defined on all types. And we can use mixins to get more mileage out of this. As it stands there's no real distinction between values of `Digital`, `Real`, `Elect`, etc. in terms of the methods you can call on them and what they return. Basically, I'm suggesting that sub-typing should be used to extend/override specific functionality, rather than just to convey semantics.

sgherbst, 2020-04-04:
Hierarchy looks good; only thing I would recommend is to put ``Real`` and ``Elect`` under a type called ``Analog``.  ``Real`` and ``Elect`` both represent analog signals, but use different simulator features -- ``Real`` is Verilog ``real`` type (i.e., event-driven) whereas ``Elect`` is a true SPICE net (i.e., variable in diff. eq. solver).  If a user specifies only that a signal is ``Analog``, that could resolve to ``Real`` or ``Elect`` depending on defaults for a certain kind of **fault** target, user preferences, overrides, etc.

I also thought it might be interesting to highlight a few ways that typing is used in **fault** with regard to analog signals.  (Since I'm not too familiar with the type system, some of these may not be the "right" way...)

Example 1: fault/system_verilog_target.py lines 550-551.  A type is checked to see if it is a subclass of ``RealType`` in ``generate_port_code(..., name, type_, ...)``
```
if issubclass(type_, RealType):
    t = "real"
```

Example 2: fault/spice_target lines 392-394.  A circuit port is checked to see if it is an instance of ``RealType``.  (I gather that ``type_`` in the previous example was a ``class`` whereas a ``port`` is an instance of a class?)
```
port = self.circuit.interface.ports[port_name]
if isinstance(port, (m.Bit, fault.RealType, fault.ElectType)):
    retval += [f'{port}']
```

Example 3: fault/actions.py lines 103-106.  A port is checked to see if it is an instance of Real/Elect In/Out/InOut.  Based on the previous example, maybe ``isinstance(self.port, (RealType, ElectType))`` would work & be more generic?
```
@property
def real_number_port(self):
    return isinstance(self.port, (RealIn, RealOut, RealInOut,
                      ElectIn, ElectOut, ElectInOut))

```

Example 4: fault/verilogams.py lines 69-77.  A ``type_`` is checked to see if it refers to ``ElectIn``/``ElectOut``/etc.  Based on example 1, maybe ``issubclass(type_, ElectType)`` would work and be more generic?
```
# determine port type
if type_ is ElectIn or type_ is ElectOut:
    vams_type = 'electrical'
elif type_ is RealIn or type_ is RealOut:
    vams_type = 'wreal'
elif len(type_) == 1:
    vams_type = 'wire'
else:
    vams_type = f'wire [{len(type_) - 1}:0]'
```