leonardt/fault #306: Adding task to fault

Kuree, 2021-04-26:
I'm in the process of rewriting tbg to support more complex configuration semantics. One challenging part is to dynamically select signals based on the signal values. It's typically done via driver tasks in SV-based testbench.

Most of the logic are already there, such as set values (poke), loop, and if. I think the only additional feature is how to create a task definition, in particular, input and output variables.

leonardt, 2021-04-26:
Could we do it similar to how we handle loops/if? E.g. `task1 = tester.create_task()` and task1 provides the tester interface for doing the actions? Seems like we could track a list of tasks and generate code for them pretty easily.  I'm guessing we need to add code to start the tasks and/or hook them up to the DUT?

Kuree, 2021-04-27:
> Could we do it similar to how we handle loops/if? E.g. task1 = tester.create_task() and task1 provides the tester interface for doing the actions? Seems like we could track a list of tasks and generate code for them pretty easily.

Yes this is what I have in mind.

> I'm guessing we need to add code to start the tasks and/or hook them up to the DUT?
Calling task is the same as calling a function. I think the pysv integration already added function call code gen, so that should be easily extended.

If we only focus on TB-side tasks, i.e. tasks used to drive DUT interface, I think we can just directly generate it in the top scope, which should work out just fine. There are several things to consider, however:
1. Usually there are timing control inside a task, e.g. wait for next clock cycle etc. We definitely should support that. Ideally it should work with loops/if as well, i.e. control statement inside a task.
2. Tasks in Verilator driver can be directly generated as cpp functions, where the first argument is the DUT pointer. I think this is the easiest way to implement.

Please let me know what you think.

leonardt, 2021-04-28:
That all sounds reasonable.  Do the timing controls usually use the SV wait statement?

Kuree, 2021-04-28:
Usually it's things like
```SystemVerilog
@(posedge clk);
// or
@(value);
// or event
-> event_var;
// and then
@(event_var)
```

`@` is more commonly used for clock and events, whereas `wait` is used for expressions.