leonardt/fault #309: Cant install faul.

Arka2BR, 2021-06-30:
When I try to use  "pip install fault", I get this error,

ERROR: Command errored out with exit status 1:
     command: 'E:\Anaconda\python.exe' -c 'import sys, setuptools, tokenize; sys.argv[0] = '"'"'C:\\Users\\arkap\\AppData\\Local\\Temp\\pip-install-04cqvv0m\\coreir\\setup.py'"'"'; __file__='"'"'C:\\Users\\arkap\\AppData\\Local\\Temp\\pip-install-04cqvv0m\\coreir\\setup.py'"'"';f=getattr(tokenize, '"'"'open'"'"', open)(__file__);code=f.read().replace('"'"'\r\n'"'"', '"'"'\n'"'"');f.close();exec(compile(code, __file__, '"'"'exec'"'"'))' egg_info --egg-base 'C:\Users\arkap\AppData\Local\Temp\pip-pip-egg-info-xo4m01hy'
         cwd: C:\Users\arkap\AppData\Local\Temp\pip-install-04cqvv0m\coreir\
    Complete output (5 lines):
    Traceback (most recent call last):
      File "<string>", line 1, in <module>
      File "C:\Users\arkap\AppData\Local\Temp\pip-install-04cqvv0m\coreir\setup.py", line 24, in <module>
        raise NotImplementedError(_system)
    NotImplementedError: Windows
    ----------------------------------------
ERROR: Command errored out with exit status 1: python setup.py egg_info Check the logs for full command output.

Any suggestions what to do?

rsetaluri, 2021-06-30:
fault depends on [coreir](https://github.com/rdaly525/coreir) and [pycoreir](https://github.com/leonardt/pycoreir) which contain binary distributions of C++ code. Unfortunately, this code has not been tested on Windows.

The easiest route would be to try running on a machine (or VM) with macOS or Linux.

If you do not have access to either and you are familiar with building C++ on Windows, you might be able to build CoreIR from source.

Arka2BR, 2021-06-30:
I see. I'll try it out.

leonardt, 2021-06-30:
Yes, if you'd like to pursue using this on windows native, we'll need to start with buildling coreir succesfully on that platform, so I would suggest opening an issue in that repository and we can explore what this might involve.  However using linux or macOS is likely to be much less painful given we've extensively tested on both these platforms.

Arka2BR, 2021-06-30:
Hmm I dont know if you could build it for Windows in a few hours if I raise the issue. Ill try VM for the time being...