leonardt/fault #327: Fault breaks with MagmaProtocol outputs

cdonovick, 2023-08-01:
Fault breaks when a protocol is used an output type on account of missing `debug_name` and `len`

See: #326

leonardt, 2023-08-02:
One option: have fault use `getattr` and default to `name` if no `debug name`
Another: have protocol pass through the `_get_magma_value_` debug name
Third: have fault check if magmaprotocol, if so get the underlying debug name

I feel like third might be the simplest option, but perhaps we can come up with some clever way to implicitly pass through these methods to the underlying magma value, or do we like having the user dispatch on this explicitly?

cdonovick, 2023-08-02:
Id prefer not to have write a `debug_name` method. In my ideal world lib code would call `proto._get_magma_value_().debug_name` or do option 1, I have no preference between the two.

Adding `debug_name` as a mixin to the `MagmaProtocol` is fine but each mixin method makes  the protocol slightly harder to use because I need to make sure I dont alias a name reserved by Magma.